SpectacularAI/sdk-examples #1: I installed the tutorial on win10 and run the pen_3d demo

qwe5788319, 2021-11-02:
C:/Users/Administrator/Desktop/sdk-examples/python/oak/pen_3d.py:18: DeprecationWarning: setRectifyMirrorFrame() is deprecated.
  vio_pipeline = spectacularAI.depthai.Pipeline(pipeline)[](url)

rgb pictures only rgb, no pen.

qwe5788319, 2021-11-02:
[image]
#

qwe5788319, 2021-11-02:
[image]
Is my environment version wrong?

oseiskar, 2021-11-02:
Hi Andy!

In the pen example, the "ink" activates when you completely cover the RGB camera with your finger (so that the "rgb" image looks completely black or very dark). Did you try that?

To verify that your setup is correct, also try the `vio_visu.py` example. It's otherwise like the pen example but the "ink" is always active and you do not have to cover the color camera.

19920716, 2021-11-02:
Hi @oseiskar 
I tried it, there is no drawing function, mixed_reality.py runs normally, but the FPS is a bit low.

Bercon, 2021-11-04:
Hi @19920716, I tried the example with same dependency versions as you have on Windows 10. The example seems to work fine despite giving the deprecation warning.
* Do you see the orange dot that marks the current position of the pen/device?
* When covering the RGB middle camera, it has to be really well covered, showing an almost completely black image to properly trigger the ink function and leaving a trail

closed 2022-03-28 by Bercon