SpectacularAI/sdk-examples #16: Low USB Speed Error with PoE

MadlyFX, 2022-03-25:
Hi, using an Oak-D PoE, I can't get any of the examples to run. Getting this error:

```
RuntimeError: Spectacular AI SDK:  error: LOW USB SPEED!

############################################################################
Your USB connection speed 'UNKNOWN' is lower than required.
Please check that you have connected the device to an USB 3 port with an
undamaged USB 3 cable (blue interior). Speed SUPER or better is recommended.
############################################################################

```

I can get image and run other (non-Spectacular) scripts so the camera is definitely visible to the network.

Bercon, 2022-03-25:
We haven't (yet) tested with Oak-D PoE, but it should work. You can disable the USB speed check like this:
```
config = spectacularAI.depthai.Configuration()
config.ensureSufficientUsbSpeed = false
vio_pipeline = spectacularAI.depthai.Pipeline(pipeline, config)
```

closed 2022-03-25 by MadlyFX

Bercon, 2022-03-28:
I finally had a chance to test Oak-D PoE and by disabling the `ensureSufficientUsbSpeed` check as mentioned above I got it working without any other changes.

We'll fix this in the next version, so you'll just see the warning that the bandwidth may be limited, but it wont fail on it by default.