SpectacularAI/sdk-examples #20: Cannot download spectacularAI from Pip

VilobatirKedis, 2022-05-26:
I can't download the spectacularAI SDK from Pip. It gives me this error

`(pi) pi@pi:~$ pip3 install spectacularAI
ERROR: Could not find a version that satisfies the requirement spectacularAI (from versions: none)
ERROR: No matching distribution found for spectacularAI
`
Pip3 debug

`(pi) pi@pi:~$ pip3 debug -v
Traceback (most recent call last):
  File "/home/pi/bin/pip3", line 8, in <module>
    sys.exit(main())
  File "/home/pi/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 73, in main
    command = create_command(cmd_name, isolated=("--isolated" in cmd_args))
  File "/home/pi/lib/python3.8/site-packages/pip/_internal/commands/__init__.py", line 96, in create_command
    module = importlib.import_module(module_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/lib/python3.8/site-packages/pip/_internal/commands/debug.py", line 11, in <module>
    from pip._vendor.certifi import where
ModuleNotFoundError: No module named 'pip._vendor.certifi'
`

oseiskar, 2022-05-27:
Note that the PyPI version is only available for 64-bit x86 Windows and Linux (but not, e.g., ARM/RPi or macOS)

VilobatirKedis, 2022-05-27:
> 

So how can I install it on RPi? I am using Ubuntu mate 20.04 on a Raspberry Pi 4

sxj731533730, 2022-05-28:
pi@raspberrypi:~/sxj731533730 $ pip3 install spectacularAI
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement spectacularAI
ERROR: No matching distribution found for spectacularAI

i still have not acquire spectacularAI  in  raspberry pi 4。。。。 
我在树莓派4上也没法下载这个pip包 -spectacularAI

oseiskar, 2022-05-30:
You can contact us at https://www.spectacularai.com/#contact for commercial licenses to the RPi version

closed 2022-05-30 by oseiskar