SpectacularAI/sdk-examples #27: mapping_visu.py not runing

leizhao3, 2022-06-25:
I'm not able to run mapping_visu.py with the following error. 
- Python 3.9.13
- Windows
- USB speed low but sufficient to run vio_jsonl.py, vio_visu.pypen_3d.py.

```
python .\mapping_visu.py
Starting OAK-D device
Close the window to stop mapping
SpectacularAI DEBUG: auto-scaling camera calibration parameters with factor 0.5
SpectacularAI DEBUG: starting SLAM thread
Spectacular AI SDK: WARN: LOW USB SPEED!

############################################################################
Your USB connection speed 'HIGH' is lower than required.
Please check that you have connected the device to an USB 3 port with an
undamaged USB 3 cable (blue interior). Speed SUPER or better is recommended.
Disable this check by setting config.ensureSufficientUsbSpeed to false.
############################################################################

SpectacularAI DEBUG: swizzle input colors as rrr (color format 1)
SpectacularAI DEBUG: applying IMU stationarity ZUPT
SpectacularAI DEBUG: First time init failed to track in 3.000000 seconds, reseting
SpectacularAI DEBUG: quitting SLAM...
SpectacularAI DEBUG: signaling the SLAM thead to quit
SpectacularAI DEBUG: starting SLAM thread
SpectacularAI DEBUG: waiting for the SLAM thead to quit
SpectacularAI DEBUG: ~Worker end
SpectacularAI DEBUG: ... SLAM finished
SpectacularAI DEBUG: applying IMU stationarity ZUPT
SpectacularAI DEBUG: First time init failed to track in 3.000000 seconds, reseting
SpectacularAI DEBUG: quitting SLAM...
SpectacularAI DEBUG: signaling the SLAM thead to quit
SpectacularAI DEBUG: starting SLAM thread
SpectacularAI DEBUG: waiting for the SLAM thead to quit
SpectacularAI DEBUG: ~Worker end
SpectacularAI DEBUG: ... SLAM finished
SpectacularAI DEBUG: applying IMU stationarity ZUPT
SpectacularAI DEBUG: First time init failed to track in 3.000000 seconds, reseting
SpectacularAI DEBUG: quitting SLAM...
SpectacularAI DEBUG: signaling the SLAM thead to quit
SpectacularAI DEBUG: waiting for the SLAM thead to quit
SpectacularAI DEBUG: starting SLAM thread
SpectacularAI DEBUG: ~Worker end
SpectacularAI DEBUG: ... SLAM finished
SpectacularAI DEBUG: applying IMU stationarity ZUPT
SpectacularAI DEBUG: IMU stationarity release update
SpectacularAI DEBUG: applying IMU stationarity ZUPT
SpectacularAI DEBUG: First time init failed to track in 3.000000 seconds, reseting
SpectacularAI DEBUG: quitting SLAM...
SpectacularAI DEBUG: signaling the SLAM thead to quit
SpectacularAI DEBUG: waiting for the SLAM thead to quit
SpectacularAI DEBUG: starting SLAM thread
SpectacularAI DEBUG: ~Worker end
SpectacularAI DEBUG: ... SLAM finished
SpectacularAI DEBUG: applying IMU stationarity ZUPT
SpectacularAI DEBUG: IMU stationarity release update
[Open3D WARNING] GLFW Error: WGL: Failed to make context current: The requested resource is in use. 
Stack trace (most recent call last):
#26   Object "", at 00007FF8F1307034, in BaseThreadInitThunk
#25   Object "", at 00007FF8EF0D1BB2, in configthreadlocale
#24   Object "", at 00007FF898F97B5E, in PyEval_AcquireThread
#23   Object "", at 00007FF898F97BE2, in PyEval_AcquireThread
#22   Object "", at 00007FF898F6065F, in PyObject_Call
#21   Object "", at 00007FF898F6085C, in PyVectorcall_Call
#20   Object "", at 00007FF898F2BC6D, in PyObject_GC_Del
#19   Object "", at 00007FF898F2C8B2, in PyFunction_Vectorcall
#18   Object "", at 00007FF898F302A1, in PyEval_EvalFrameDefault
#17   Object "", at 00007FF898F302A1, in PyEval_EvalFrameDefault
#16   Object "", at 00007FF898F346FB, in PyEval_EvalFrameDefault
#15   Object "", at 00007FF898F6074E, in PyObject_Call
#13   Object "", at 00007FF898F2C8B2, in PyFunction_Vectorcall
#12   Object "", at 00007FF898F30C3E, in PyEval_EvalFrameDefault
#11   Object "", at 00007FF898F302A1, in PyEval_EvalFrameDefault
#10   Object "", at 00007FF898F30428, in PyEval_EvalFrameDefault
#9    Object "", at 00007FF89906E7E3, in Py_gitversion
#8    Object "", at 00007FF898F25215, in PyObject_MakeTpCall
#7    Object "", at 00007FF898F78A8A, in PyObject_Str
#6    Object "", at 00007FF836D63A39, in DeadlyExportError::~DeadlyExportError
#5    Object "", at 00007FF837046184, in librealsense::color_map::get_cache
#4    Object "", at 00007FF8370460FC, in librealsense::color_map::get_cache
#3    Object "", at 00007FF8373B44D9, in DeadlyErrorBase::operator=
#2    Object "", at 00007FF8375B7EEA, in aiGetVersionMinor
#1    Object "", at 00007FF8375AD2D5, in aiGetVersionMinor
#0    Object "", at 00007FF84F33AB4B, in DumpRegistryKeyDefinitions
```

swatty007, 2022-06-28:
Just now run into the same problem. I'm using an OAK-D-POE tough and accordingly tried to set the USB Speed warning to false. Which then results in the error shown above.

closed 2022-07-04 by kaatrasa

swatty007, 2022-07-04:
Perfect, seems to work with the OAK-D-POE now :) Thanks for the quick fix.

PR #29 (2022-07-04): Fix mapping_visu example on Windows
Open3D visualizer functions are now only called from the main(open3d) thread. Should fix #27