SpectacularAI/sdk-examples #34: Not able run vio_mapper in Linux_ubuntu_x87-64

woshicaijihahahaha, 2022-08-22:
hello,when i was running  ./vio_mapper /home/yy/EDisk/recording/

then some wrong messages occured :
SpectacularAI WARN: Unknown camera model RS2_DISTORTION_INVERSE_BROWN_CONRADY, ignoring distortion coefficients.
SpectacularAI WARN: Unknown camera model RS2_DISTORTION_BROWN_CONRADY, ignoring distortion coefficients.
terminate called after throwing an instance of 'std::runtime_error'
  what():  unused key(s): icpMinOverlapRatio

so how can i fix it please.

pekkaran, 2022-08-22:
Please remove from `recording/vio_config.yaml` the line that starts with `icpMinOverlapRatio:`.

The warnings about distortion can be ignored, the RealSense video inputs are rectified or closely so and the default pinhole model without distortion should work well.

closed 2022-08-22 by pekkaran

woshicaijihahahaha, 2022-08-23:
Hello ，I found the following in the **vio_config.yaml file**：

noiseScale: 500
trackMinFrames: 3
trackChiTestOutlierR: 3
trackOutlierThresholdGrowthFactor: 1.3
hybridMapSize: 2
sampleSyncLag: 1
estimateImuCameraTimeShift:  
outputCameraPose: True
imuStationaryEnabled: True
trackingStatusInitMinSeconds: 0.5
useGnss: true
imuStationaryEnabled: False
trackingStatusInitMinSeconds: 2.0
skipFirstNCandidates: 10
noiseProcessBGA: 1e-5
visualR: 0.01
cameraTrailLength: 12
cameraTrailHanoiLength: 8

I didn't find the *icpMinOverlapRatio* you mentioned,I am still a newbie in these areas，Can you look at it again for me?

pekkaran, 2022-08-23:
Sorry, I misunderstood your issue. The problem has been fixed in the [1.3.0 release](https://github.com/SpectacularAI/sdk/releases). Could you try again with that version?