SpectacularAI/sdk-examples #40: What is "spectacularAI_realsensePlugin"? How to install that?

JimXu1989, 2022-10-28:
Thanks for the Great sharing
I met a Error when I try to build the cpp examples:
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "FindspectacularAI_realsensePlugin.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "spectacularAI_realsensePlugin", but CMake
  did not find one.

  Could not find a package configuration file provided by
  "spectacularAI_realsensePlugin" with any of the following names:

    spectacularAI_realsensePluginConfig.cmake
    spectacularai_realsenseplugin-config.cmake

  Add the installation prefix of "spectacularAI_realsensePlugin" to
  CMAKE_PREFIX_PATH or set "spectacularAI_realsensePlugin_DIR" to a directory
  containing one of the above files.  If "spectacularAI_realsensePlugin"
  provides a separate development package or SDK, be sure it has been
  installed.

Configuring incomplete, errors occurred!
See also "/home/xss/SoftWare/sdk-examples/cpp/realsense/build/CMakeFiles/CMakeOutput.log".

What is "spectacularAI_realsensePlugin"? How to install that?
Thanks a lot

oseiskar, 2022-10-28:
You can find it here https://github.com/SpectacularAI/sdk/releases

JimXu1989, 2022-10-29:
> You can find it here https://github.com/SpectacularAI/sdk/releases

Hi, thanks for your reply, I found the realsense plugin, but when I build the cpp examples, I met another error, I have install realsense 2.51 on my computer, but seems realsense plugin need 2.49, should I install 2.49 instead?:
[100%] Linking CXX executable vio_mapper
/usr/bin/ld: warning: librealsense2.so.2.49, needed by /usr/local/lib/libspectacularAI_realsensePlugin.so.1.7.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: CMakeFiles/vio_mapper.dir/vio_mapper.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/vio_mapper.dir/build.make:168：vio_mapper] 错误 1
make[1]: *** [CMakeFiles/Makefile2:113：CMakeFiles/vio_mapper.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
/usr/bin/ld: warning: librealsense2.so.2.49, needed by /usr/local/lib/libspectacularAI_realsensePlugin.so.1.7.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: CMakeFiles/vio_mapper_legacy.dir/vio_mapper_legacy.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/vio_mapper_legacy.dir/build.make:168：vio_mapper_legacy] 错误 1
make[1]: *** [CMakeFiles/Makefile2:139：CMakeFiles/vio_mapper_legacy.dir/all] 错误 2
make: *** [Makefile:91：all] 错误 2

JimXu1989, 2022-10-29:
> You can find it here https://github.com/SpectacularAI/sdk/releases

When I copy realsense2.so.2.49 to usr/local/lib, I met another problem:
[100%] Linking CXX executable vio_mapper
/usr/bin/ld: warning: librealsense2.so.2.49, needed by /usr/local/lib/libspectacularAI_realsensePlugin.so.1.7.0, may conflict with librealsense2.so.2.51
/usr/bin/ld: CMakeFiles/vio_mapper_legacy.dir/vio_mapper_legacy.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/vio_mapper_legacy.dir/build.make:168：vio_mapper_legacy] 错误 1
make[1]: *** [CMakeFiles/Makefile2:139：CMakeFiles/vio_mapper_legacy.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
/usr/bin/ld: warning: librealsense2.so.2.49, needed by /usr/local/lib/libspectacularAI_realsensePlugin.so.1.7.0, may conflict with librealsense2.so.2.51
/usr/bin/ld: CMakeFiles/vio_mapper.dir/vio_mapper.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/vio_mapper.dir/build.make:168：vio_mapper] 错误 1
make[1]: *** [CMakeFiles/Makefile2:113：CMakeFiles/vio_mapper.dir/all] 错误 2
make: *** [Makefile:91：all] 错误 2

JimXu1989, 2022-10-29:
Hi, I have successfully installed the examples,
The vio_jsonl I built seems need wait a couple second before it get work.
when I run the other two, it show:
Usage: ./vio_mapper /path/to/recording/folder
double free or corruption (!prev)
已放弃 (核心已转储)

Usage: ./vio_mapper_legacy /path/to/recording/folder [N]
where N is the frame sampling interval, default: 10
double free or corruption (!prev)
已放弃 (核心已转储)

Could you show me how to use it?

pekkaran, 2022-11-01:
Hello,

Both the binaries are used like this:
```
mkdir recording
./vio_mapper recording
```

If you use the latest example code version that includes this [commit](https://github.com/SpectacularAI/sdk-examples/commit/fa654add4aa6bf62d139ab7fdff55b7829dc7d7e), then creating the directory manually is not required.

If you still see the error `double free or corruption (!prev)`, please let us know. That should not happen.

closed 2022-11-29 by oseiskar