SpectacularAI/sdk-examples #56: [Request] Help with debugging a simple example with Luxonis OAK RVC3 devices

moratom, 2023-02-07:
Hello SpectularAI team!

I am working on getting a simple example to run on our new series (RVC3) device(s) and would greatly appreciate some help :)

For now I have a simple request: 
could you tell me **under which condition** this exception is thrown in your library?
```
(envRVC3) matevz@matevz-PC:~/Luxonis/Testing/$ python vio_jsonl.py 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Spectacular AI SDK:  error: unexpected dimensions in depth image
```

The example I'm trying to run is attached [here](https://github.com/SpectacularAI/sdk-examples/files/10670337/vio_jsonl.zip), though likely not very relevant, since the issue is likely on our side.

oseiskar, 2023-02-08:
Hi @moratom . Which RVC3 device are you using?

We do not have out-of-the-box support for the multi-camera devices our Depth AI plugin yet. If the problem is related to a multi-camera device, you need to ensure that you have two identical camera modules attached to both the "LEFT" and "RIGHT" sockets (which may be aliases of "CAM_B" and "CAM_C" on multi-camera devices, see [here](https://docs.luxonis.com/projects/hardware/en/latest/pages/DD2090.html)). Those should also be monocular global shutter cameras with native resolution 1280x800.

If this works, manual camera calibration (see [here](https://spectacularai.github.io/docs/pdf/oak_fisheye_calibration_instructions.pdf)) and specifying the IMU-to-camera matrix (see [here](https://github.com/SpectacularAI/sdk-examples/tree/main/python/oak#unsupported-oak-d-models)) are also needed.

The following configuration flags may also help debugging:
 * [`useStereo`](https://spectacularai.github.io/docs/sdk/python/latest/#spectacularAI.depthai.Configuration.useStereo): Set to false to try if it works with a single camera
 * [`useFeatureTracker`](https://spectacularai.github.io/docs/sdk/python/latest/#spectacularAI.depthai.Configuration.useFeatureTracker). Set to false to disable VPU-acceleration to see if the problem is related to that

moratom, 2023-02-13:
We are using FFC-6P with OV9282 on the RIGHT and LEFT sockets, so this shouldn't be a problem.
The culprit was highly likely the resolution, since we didn't have support for the 800p resolution on the OV9282 on RVC3 cameras up until very recently and the mono cameras were working at 720p.

Thanks a lot. I'll try debug this further in the following days and hopefully get it working.

closed 2023-11-18 by oseiskar