SpectacularAI/sdk-examples #61: Loading existing map

kekeblom, 2023-02-25:
I tried saving a map using the `mapSavePath` configuration parameter and then starting a new session with the `mapLoadPath` parameter set. Depending on the size of the map, it either crashes at startup with the error:
```
Starting OAK-D device                                                       
SpectacularAI WARN: missing key frame in pose trail, skipping VIO constraint
terminate called after throwing an instance of 'std::out_of_range'          
  what():  map::at                                                          
``` 
or it just hangs indefinitely with a larger map. 

The device used is an OAK-D Pro and I tried both SDK versions 1.12.1 and 1.13.1. Let me know if I can assist in some way to help figure out what is going on here. 

On another note, would it be possible to get the current camera pose in map frame instead of mission frame? Would be nice to be able to have the coordinate frames be consistent across several launches.

oseiskar, 2023-03-08:
Sorry for the slow answer and thanks for reporting. We'll take a look...

kaatrasa, 2023-03-13:
Hey,

Could you post the commands you used to produce the error? For me, the following works:
`python vio_record.py --map --output test`
And then
`python mixed_reality.py --mapLoadPath test/slam_map.bin`
with most recent release (v1.14)?

kekeblom, 2023-05-08:
Dropped the ball on this one.

The commands are:
```
python mapping_ros.py --save-map /tmp/map.bin
python mapping_ros.py --map /tmp/map.bin
```

The error I get is the same one as above.
```
Starting OAK-D device
SpectacularAI WARN: missing key frame in pose trail, skipping VIO constraint
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
```

kaatrasa, 2023-05-31:
Hey, sorry for taking so long!

I tried again, but couldn't still reproduce the issue using `vio_record.py` and `mixed_reality.py` (I would like to avoid installing ROS for `mapping_ros.py`) using the exact same `config.internalParameters` your `mapping_ros.py` example uses.

Just making sure that you used the same SDK version for `python mapping_ros.py --save-map /tmp/map.bin` and `python mapping_ros.py --map /tmp/map.bin`? The `.bin` format is not compatible between different SDK versions... we will add automatic detection for this in future versions of the SDK. 

I can also take a look at the issue if you could share me the `map.bin` and a short dataset `test` that tries to localize in the map:
`python mapping_ros.py --save-map /tmp/map.bin`
`python vio_record.py --output test`

oseiskar, 2023-09-15:
Reopen if the issue persists

closed 2023-09-15 by oseiskar

framacia, 2023-11-29:
Hey, I have experienced the same issue, recorded a map using `vio_record.py --map`, then tried to load it in with `mixed_reality.py --mapLoadPath`. It starts running correctly (but with the default cube not rendered), and then a soon as the camera moves it crashes. Here is the stack trace

```
SpectacularAI ERROR: failed to open /data/slam_map.bin for reading
Stack trace (most recent call last) in thread 11990:
#9    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#8    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f322401d132, in clone
#7    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f3223ee3608, in 
#6    Object "/home/fubintlab/.local/lib/python3.8/site-packages/spectacularAI.cpython-38-x86_64-linux-gnu.so", at 0x7f3217a0d41f, in 
#5    Object "/home/fubintlab/.local/lib/python3.8/site-packages/spectacularAI.cpython-38-x86_64-linux-gnu.so", at 0x7f321698aac5, in 
#4    Object "/home/fubintlab/.local/lib/python3.8/site-packages/spectacularAI.cpython-38-x86_64-linux-gnu.so", at 0x7f32169938ed, in 
#3    Object "/home/fubintlab/.local/lib/python3.8/site-packages/spectacularAI.cpython-38-x86_64-linux-gnu.so", at 0x7f3216993495, in 
#2    Object "/home/fubintlab/.local/lib/python3.8/site-packages/spectacularAI.cpython-38-x86_64-linux-gnu.so", at 0x7f32166f4ebe, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3223f20858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3223f4100b, in gsignal
```