SpectacularAI/sdk-examples #62: Realigning after tracking loss

gseqBE, 2023-03-08:
Hello, 
We are using the OakD with SDK 1.14.

Is there a way(either by saving or by setting the last co-ordinates before tracking loss) to correct the camera pose.
We are seeing huge jumps, sometimes back to 0,0 when the SpecAI reports a tracking loss.

Thank you!

oseiskar, 2023-03-09:
Hello. Currently we do not have this relocalization feature (after tracking loss) in the SDK and there are no immediate plans to implement it, unfortunately.

gseqBE, 2023-03-09:
@oseiskar 
Could you explaing what happens to the pose, once tacking recovers from a loss? If I can understand the recovery model, I can manage it on the host side.
Thank you.

oseiskar, 2023-03-09:
We do not specify/guarantee what exactly happens in this case. It is implementation specific and subject to change between SDK versions. In practice, the SDK currently tries to restart from a "last good known position", but this may have large errors depending on how the tracking fails and what happens just prior to that. To be on the safe side, I would recommend to handle each contiguous sequence of poses in TRACKING states (after startup or a LOST state) as if the initial position and heading were arbitrary/random.

closed 2023-03-09 by gseqBE