SpectacularAI/sdk-examples #72: vio_jsonl.cpp runtime exception when built in Debug mode

SClerckx, 2023-05-08:
Hi SpectacularAI Team!

I'm currently trying to build vio_jsonl.cpp in Debug mode to integrate it with another project. The building is successful however I encounter a runtime exception:

`Unhandled exception at 0x00007FF9DCCEFDEC in vio_jsonl.exe: Microsoft C++ exception: std::bad_alloc at memory location 0x0000000F285DE430.`

This is printed in the terminal before the exception:
`[warning] USB protocol not available - If running in a container, make sure that the following is set: "-v /dev/bus/usb:/dev/bus/usb --device-cgroup-rule='c 189:* rmw'"`

This is the stack trace:
```
KernelBase.dll!00007ff9dccefdec()
vcruntime140.dll!00007ff9acee6720()
spectacularAI_depthaiPlugin.dll!00007ff8b7714a5f()
spectacularAI_depthaiPlugin.dll!00007ff8b80fb169()
spectacularAI_depthaiPlugin.dll!00007ff8b76d80be()
spectacularAI_depthaiPlugin.dll!00007ff8b76d853f()
spectacularAI_depthaiPlugin.dll!00007ff8b76dec7a()
spectacularAI_depthaiPlugin.dll!00007ff8b76debfc()
vio_jsonl.exe!main(int argc, char * * argv) Line 21
[External Code]
```
This is in Visual Studio 2019, however, the same issue occurs when following the [readme step in Powershell](https://github.com/SpectacularAI/sdk-examples/tree/main/cpp/oak). 
The Release version works fine both in Visual Studio and Powershell. I tried placing the non-debug libusb-1.0.dll with the vio_jsonl.exe. This removed the print USB protocol statement before the exception but doesn't seem to fix anything else.

Bercon, 2023-05-10:
Hi,

We don't make the debug binaries available to public, so you will need to debug in Release mode. This may help you https://learn.microsoft.com/en-us/cpp/build/how-to-debug-a-release-build?view=msvc-170

SClerckx, 2023-05-10:
Okay, thank you

closed 2023-05-10 by SClerckx