SpectacularAI/sdk-examples #87: Spectacular SDK mapping_visu.py is dropping frames of OAK-D

ostepan2006, 2023-09-15:
Hello Team. I had been tryng Spectacular SDK `mapping_visu.py` with OAK-D and got tons of messages "SpectacularAI : dropping frames! VIO is running too slow or data is being input too fast".  What does this mean? I had been using Asus notebook with Core i5 and 4 Gb RAM. Might this mean that capability of my notebook are poor for Spectacular SDK examples? 
What can I do to run working Spectacular SDK examples?

Thanks.

ostepan2006, 2023-09-15:
Hi Team. I had resolved the issue. Thanks. I would had been deleted this request but there is no oppurtunitie.

closed 2023-09-15 by oseiskar

Contain1R, 2023-11-21:
> 大家好。我已经解决了这个问题。谢谢。我本来会删除这个请求，但没有机会。

怎么解决的呀

Contain1R, 2023-11-21:
> Hi Team. I had resolved the issue. Thanks. I would had been deleted this request but there is no oppurtunitie.

how to resolved？

ajsnyde, 2024-03-06:
@ostepan2006 
How did you resolve it???