SpectacularAI/sdk-examples #91: Using AprilTags returns error

mgagvani, 2023-11-05:
When I add an AprilTags JSON file to the config I get this error:
```
[184430103199860E00] [1.1] [1.990] [Script(9)] [critical] AttributeError: 'lpb.ImgFrame' object has no attribute 'getTimestampDevice'

At:
  <script>(10): <module>

F: [global] [         0] [ThreadN] dispatcherResponseServe:925  no request for this response: XLINK_READ_REL_RESP 1
```
If I comment out the `config.aprilTagPath` part the code will run.

I am using an Oak-D Pro camera. My JSON configuration file is below:
```json
[
    {
    "id": 0,
    "size": 0.152,
    "family": "tag16h5",
    "tagToWorld": [
        [0, 0, 1, 0],
        [0, 1, 0, 1.494],
        [-1, 0, 0, -0.321],
        [0, 0, 0, 1]]
    },
    {
        "id": 1,
        "size": 0.152,
        "family": "tag16h5",
        "tagToWorld": [
            [1, 0, 0, 1.735],
            [0, 1, 0, 2.058],
            [0, 0, 1, -0.435],
            [0, 0, 0, 1]]
    }
]
```
Any help would be appreciated.

kaatrasa, 2023-11-10:
Hey,

It doesn't look like the error messages should be related to AprilTags, but if setting `aprilTagPath` is causing some issues, then it would be good to verify that your apriltag config file is correct. I would double-check the `tagToWorld` transformations that are a little tricky to get right.

[image]
For instance, is your second tag attached to the ceiling? We use z-is-up world coordinate system (`z = -gravity`), and now you have `tag.z == world.z`. To get started, you could try with a single AprilTag whose y-axis is pointing towards gravity direction and use this transformation:
```
"tagToWorld": [
[1, 0, 0, 0],
[0, 0, 1, 0],
[0, -1, 0, 0],
[0, 0, 0, 1]]
```

These resources might be useful:
https://github.com/SpectacularAI/sdk-examples/tree/main/python/oak#coordinate-systems
https://github.com/SpectacularAI/docs/blob/main/pdf/april_tag_instructions.pdf

If the issues persist, then maybe the easiest option would be for you to share a recording with us, see [vio_record.py](https://github.com/SpectacularAI/sdk-examples/blob/main/python/oak/vio_record.py).

mgagvani, 2023-11-11:
Thank you. The tag to world transforms are a bit confusing to me. I see the example you gave in your comment and in the apriltag instructions PDF but I don't understand where I am supposed to input the actual ground truth position of the AprilTags in this. My tagToWorld looks like this:
```json
"tagToWorld": [
        [0, 0, 1, 0],
        [0, 1, 0, 1.494],
        [-1, 0, 0, -0.321],
        [0, 0, 0, 1]]
```
This apriltag is pointed horizontally on a wall, and from my understanding, since Z goes into the AprilTag, I flipped that number in the matrix. And then the right-hand column is the x,y,z translation of the apriltag from the origin. If the configuration is like how you specified above where should I be putting the actual position of the apriltag?

kaatrasa, 2023-11-13:
tagToWorld is a transformation matrix like this:

$$\begin{bmatrix}
    \mathbf{R} & \mathbf{t} \\
    \mathbf{0} & 1
\end{bmatrix}$$

where $\mathbf{R}$ is a 3x3 rotation matrix and 

$$
\mathbf{t} = 
\begin{bmatrix} 
  tx \\
  ty \\ 
  tz 
\end{bmatrix}$$ 

is 3x1 translation vector, which defines the tag's position in world coordinate frame. The rotation part is tricky to get right without visualization (we will try to add one eventually). If you are unsure about it, I would suggest you place the tag on a wall, keep its original orientation and use the rotation matrix

$$
\mathbf{R} = 
\begin{bmatrix} 
  1 && 0 && 0 \\
  0 && 0 && 1 \\ 
  0 && -1 && 0
\end{bmatrix}$$

If you have more AprilTags then you could try virtually placing the tags in some 3D visualization (e.g. Blender), and exporting the transformations from there.

mgagvani, 2023-11-19:
Thanks for your input @kaatrasa. I understand the tagToWorld transformation but I am receiving a low-level error from the camera whenever I add this to my code:
```python
config = spectacularAI.depthai.Configuration()
config.aprilTagPath = "test_apriltags.json"
pipeline = depthai.Pipeline()
vio_pipeline = spectacularAI.depthai.Pipeline(pipeline, config)
```
The error is the original issue: 

> [184430103199860E00] [1.1] [1.990] [Script(9)] [critical] AttributeError: 'lpb.ImgFrame' object has no attribute 'getTimestampDevice'

> At:
>  <script>(10): <module>

> F: [global] [         0] [ThreadN] dispatcherResponseServe:925  no request for this response: XLINK_READ_REL_RESP 1

If the line setting `config.aprilTagPath` is commented out, this is sufficient to prevent the error. I am asking what could be the issue which is causing this particular behaviour?

kaatrasa, 2023-11-20:
Thanks for the debugging help! It indeed does seem like an issue in our end. I will try to reproduce it and let you know when a fix is available (next release or patch release).

Bercon, 2023-11-20:
Do you have the latest version of depthai and spectacularAI? You can ensure this by reinstalling both:
```
pip install depthai spectacularai --force
```
Additionally, does the mapping_visu.py work for you? Just run following from the `python/oak` examples and you should be able to create a point cloud of your surroundings:
```
python vio_mapping.py
```

mgagvani, 2023-11-20:
Yes, the mapping_visu.py does work for me. 

Okay, after updading to the latest version, the apriltag detector does work! Although, I am getting these warnings:
`SpectacularAI WARN: Detected AprilTag with id=2 multiple times in the same image!` and `SpectacularAI WARN: missing key frame in pose trail, skipping VIO constraint`. 

To clarify, when I recieve the keyframe warning, does this mean that the VIO is still updating? My use-case is that I will have multiple April-Tags in view for about 50% of the time and no AprilTags in view the other half. 

Also, does each AprilTag detection act as a loop closure independently, or do multiple detected AprilTags simultaneously update the known true position?

 Thank you for your help.

kaatrasa, 2023-11-21:
This warning message: `SpectacularAI WARN: Detected AprilTag with id=2 multiple times in the same image!` is saying that the SDK detected the same AprilTag twice in the same image.  The AprilTags should be unique (i.e. only use each AprilTag once). Otherwise, we cannot unambiguously compute camera world pose using the AprilTag, and tracking will likely break when it jumps between the AprilTags.

And yes, internally, the AprilTags are used for loop closing as part of our SLAM optimization. Detecting multiple AprilTags in the same image should help localization accuracy (as long as they are unique and the `tagToWorld` transformations are accurate).

I am guessing the `skipping VIO constraint` is related to the issue I explained above. And your use-case sounds good :+1:

mgagvani, 2023-11-26:
Thanks for the clarification. Here's an example of the "Detected AprilTag multiple times" error occuring: 
[image]
As you can see there's only 1 AprilTag in the image.

kaatrasa, 2023-11-27:
Ok, seems very strange then. Does that error/warning happen consistently? Maybe easiest to debug would be if you could send us a recording and I could take a quick look. To record a dataset, just run:
```
python vio_record.py --output /path/to/recording
```
To verify that the error is present in your dataset, you should probably set `config.aprilTagPath` in vio_record.py. Also, please share your AprilTag json file.

mgagvani, 2023-11-29:
Okay, this is weird. I ran the `vio_record.py` script (just pulled the code, so it should be up to date). Here is the output:
```cmd
python .\vio_record.py --output debug_apriltag
C:\Users\mgagv\Documents\FRC\sdk-examples\python\oak\vio_visu.py:100: UserWarning: frames=None which we can infer the length of, did not pass an explicit *save_count* and passed cache_frame_data=True.  To avoid a possibly unbounded cache, frame data caching has been disabled. To suppress this warning either pass `cache_frame_data=False` or `save_count=MAX_FRAMES`.
  anim = FuncAnimation(fig, update_graph, interval=15, blit=True)       
Recording!

Close the visualization window to stop recording
[184430103199860E00] [1.1] [1.680] [Script(9)] [critical] AttributeError: 'lpb.ImgFrame' object has no attribute 'getTimestampDevice'

At:
  <script>(10): <module>

F: [global] [         0] [ThreadN] dispatcherResponseServe:925  no request for this response: XLINK_READ_REL_RESP 1

Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\mgagv\AppData\Local\Programs\DepthAI\WPy64-3950\python-3.9.5.amd64\lib\threading.py", line 954, in _bootstrap_inner
    self.run()
  File "C:\Users\mgagv\AppData\Local\Programs\DepthAI\WPy64-3950\python-3.9.5.amd64\lib\threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\mgagv\Documents\FRC\sdk-examples\python\oak\vio_record.py", line 225, in <lambda>
    reader_thread = threading.Thread(target = lambda: main_loop(plotter))
  File "C:\Users\mgagv\Documents\FRC\sdk-examples\python\oak\vio_record.py", line 174, in main_loop
    if rgbQueue.has():
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'h265-rgb' (X_LINK_ERROR)'
frame=    1 fps=0.0 q=-0.0 Lsize=       4kB time=00:00:00.00 bitrate=32688.0kbits/s speed=0.00859x
```

The error is consisent regardless of whether I set `config.aprilTagPath`.

Here is the output running the script generated: https://drive.google.com/drive/folders/1tjs9S3tVOfNH3yn8g0CBsvL5Oh7S8dy2?usp=sharing

mgagvani, 2023-12-04:
When I run `python vio_record.py --output /path/to/recording --no_rgb` I get the same error: 
```bash
C:\Users\mgagv\Documents\FRC\sdk-examples\python\oak\vio_visu.py:100: UserWarning: frames=None which we can infer the length of, did not pass an explicit 
*save_count* and passed cache_frame_data=True.  To avoid a possibly unbounded cache, frame data caching has been disabled. To suppress this warning either pass `cache_frame_data=False` or `save_count=MAX_FRAMES`.
  anim = FuncAnimation(fig, update_graph, interval=15, blit=True)
[1844301081A8840E00] [1.1] [1.620] [Script(7)] [Recording!critical

] AttributeError: 'lpb.ImgFrame' object has no attribute 'getTimestampDevice'

At:

Close the visualization window to stop recording
F: [global] [         0] [ThreadN] dispatcherResponseServe:925  no request for this response: XLINK_READ_REL_RESP 1
```

However, the DepthAI RGB camera preview example does work. Here is the output: 
```python
Connected cameras: [{socket: RGB/CENTER/CAM_A, sensorName: IMX378, width: 4056, height: 3040, orientation: AUTO, supportedTypes: [COLOR], hasAutofocus: 1, name: color}, {socket: LEFT/CAM_B, sensorName: OV9282, width: 1280, height: 
800, orientation: AUTO, supportedTypes: [MONO], hasAutofocus: 0, name: left}, {socket: RIGHT/CAM_C, sensorName: OV9282, width: 1280, height: 800, orientation: AUTO, supportedTypes: [MONO], hasAutofocus: 0, name: right}]
Usb speed: SUPER
Device name: OAK-D-PRO-FF
0:00:01.362337
0:00:01.395667
```

I call `inRgb.getTimestampDevice()` in the loop which are the last two lines of the output.

kaatrasa, 2023-12-05:
Thanks again with debugging help. I think I know what is causing the issues now; when SLAM is enabled in the vioConfig then we create a script node where we pick only every nth stereo image, and for other images we create empty `ImgFrame` objects (for which we try to set timestamp only, but maybe there are some issues on some OAK-D models).

Setting `aprilTagPath` enables SLAM, and `vio_record.py `also enables SLAM; hence issues in both use cases for you. I am pretty sure you can fix the issues for now using vioConfig option `useFeatureTracker = False`, [that](https://github.com/SpectacularAI/sdk-examples/tree/main/python/oak#camera-modes) 

> disables accelerated feature tracking, which can improve accuracy, but also increases CPU consumption. Causes depth map and feature inputs to be replaced with full FPS monochrome image data.

For example, `python vio_record.py --output /path/to/recording --no_feature_tracker`

If this seems to work, then could use the `useFeatureTracker = False` mode in your AprilTag experiments until we are able to fix the issue?

mgagvani, 2023-12-11:
Yeah, I tried --no_feature_tracker and it seems to be much too CPU intensive to work at real time, although it does not crash with the error anymore. I'll leave this issue open since it still breaks when the feature tracker is on.

kaatrasa, 2023-12-21:
Just to let you know, the `getTimestampDevice` error you were getting should be now fixed in the latest release.

closed 2024-01-15 by mgagvani