SpectacularAI/sdk-examples #95: How can I access all the data points of the camera's pose positions?

kenhuang1964, 2023-11-18:
Is there a way I can access the coordinates of all the yellow pose positions? 
[image]

oseiskar, 2023-11-18:
1: Yes you can! These are the `keyFrames` in the [Mapping API output](https://spectacularai.github.io/docs/sdk/python/latest/#spectacularAI.mapping.MapperOutput). In the `replay_to_nerf.py` script, they are used like this https://github.com/SpectacularAI/sdk-examples/blob/8ef5ae723a4442a2946bf1ee117ae6a4612d2367/python/mapping/replay_to_nerf.py#L278-L294

They are also stored to disk by the same script, in a Nerfstudio-compatible format and coordinate convention (COLMAP's output format), but if you want to use them for some other purpose, it's probably easiest to modify the replay_to_nerf script where everything is already correctly parsed and accessible in a [documented coordinate convention](https://github.com/SpectacularAI/sdk-examples/tree/main/python/oak#coordinate-systems)

2: Possible, but currently experimental and not well-documented: You would first have to change the script to output the map in our internal `.bin` format (i.e., change [this line](https://github.com/SpectacularAI/sdk-examples/blob/8ef5ae723a4442a2946bf1ee117ae6a4612d2367/python/mapping/replay_to_nerf.py#L376) from `.csv` to `.bin`, which breaks the rest of the script, but gives you a localizable map), which can then be loaded by the SDK for visual positioning. (Example [here](https://github.com/SpectacularAI/sdk-examples/blob/8ef5ae723a4442a2946bf1ee117ae6a4612d2367/python/oak/mixed_reality.py#L21))

kenhuang1964, 2023-11-18:
Awesome, thank you so much @oseiskar! I love your enthusiasm :+1:

closed 2023-11-18 by kenhuang1964

kenhuang1964, 2023-11-20:
Hey @oseiskar, quick question: can I somehow get the coordinates of the keyframe within the global map? Also, is it possible to update a loaded map with more coordinates?

oseiskar, 2023-11-20:
Key frame (camera) coordinates are a part of the Mapping API output. The sparse co-visibility graph (which KFs share sparse key points) is not available in the current API