SpectacularAI/sdk-examples #98: SpectacularAI ERROR: `ffmpeg` must be installed.

orgicus, 2023-11-21:
Hi,

First off: thanks for sharing these awesome examples: really fun to play with !

I did run into an error running the vio_replay.py example: `SpectacularAI ERROR: `ffmpeg` must be installed.`

It seems it can't find the path to ffmpeg.exe ? It is added to the `PATH` environment variable and I did check python can see it too like so:

```bash
python -c "import subprocess;print(subprocess.check_output(['ffmpeg','-version']))"
b'ffmpeg version 4.2.3 Copyright (c) 2000-2020 the FFmpeg developers\r\nbuilt with gcc 9.3.1 (GCC) 20200523\r\nconfiguration: --enable-gpl --enable-version3 --enable-sdl2 --enable-fontconfig --enable-gnutls --enable-iconv --enable-libass --enable-libdav1d --enable-libbluray --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libtheora --enable-libtwolame --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libzimg --enable-lzma --enable-zlib --enable-gmp --enable-libvidstab --enable-libvorbis --enable-libvo-amrwbenc --enable-libmysofa --enable-libspeex --enable-libxvid --enable-libaom --enable-libmfx --enable-amf --enable-ffnvcodec --enable-cuvid --enable-d3d11va --enable-nvenc --enable-nvdec --enable-dxva2 --enable-avisynth --enable-libopenmpt\r\nlibavutil      56. 31.100 / 56. 31.100\r\nlibavcodec     58. 54.100 / 58. 54.100\r\nlibavformat    58. 29.100 / 58. 29.100\r\nlibavdevice    58.  8.100 / 58.  8.100\r\nlibavfilter     7. 57.100 /  7. 57.100\r\nlibswscale      5.  5.100 /  5.  5.100\r\nlibswresample   3.  5.100 /  3.  5.100\r\nlibpostproc    55.  5.100 / 55.  5.100\r\n'
```

Is there a separate env variable SpectacularAI uses to ffmpeg or a specific path ? (for example, currently I have ffmpeg installed via ImageMagick, so not a standard path?). Is there a way to tell the SpectacularAI module where to look for ffmpeg ?

oseiskar, 2023-11-21:
Thank you for reporting. I think we need to clarify the instructions / error messages around this (also ping @Bercon ). The SDK may actually use other FFMpeg command line tools besides `ffmpeg`. In particular, `ffprobe` must work too. The SDK does not try to find them in any custom paths so if typing `ffmpeg` and `ffprobe` both work on the command line where you also run the Python replay, then it _should_ work. If not, please let us know.

orgicus, 2023-11-21:
That's exactly what it was! 🎉 

I swapped environment variables away from imagemagick and to a path that also has `ffmpeg`.
For people already using 3D Gaussian Splatting a path like this may be handy:
`gaussian-splatting\SIBR_viewers\extlibs\ffmpeg\bin\ffprobe.exe`

Thank you so much @oseiskar ! 🙌

closed 2023-11-21 by orgicus