SpectacularAI/sdk-examples #100: "SpectacularAI : dropping frames! VIO is running too slow or data is being input too fast"

Contain1R, 2023-11-22:
I had been tryng Spectacular SDK mapping_visu.py with OAK-D and got tons of messages "SpectacularAI : dropping frames! VIO is running too slow or data is being input too fast". Also I got a message: IMU firmware version detected : 3.2.13 is not supported. Latest supported: 3.9.9. Use "imu.enableFirmwareUpdate(True)"once to update firmware to latest". I had update imu firmware version. But the message "SpectacularAI : dropping frames! VIO is running too slow or data is being input too fast" has not dissapeared.

ajsnyde, 2024-03-06:
Also running into this issue, minus the firmware message. I'm needing to get VIO for my application on a limited CPU system. Is there a recommended way to post-process the VIO (position, orientation, time) from a `recording-only` run?

Bercon, 2024-03-06:
You can use the Replay API (https://spectacularai.github.io/docs/sdk/replay.html). Many of our python examples also have an option of providing recorded data folder to demonstrate how to use it

You can also try enabling the "fastVio" config option to reduce CPU usage at a cost of slightly less accurate tracking https://spectacularai.github.io/docs/sdk/wrappers/oak.html#spectacularAI.depthai.Configuration.fastVio

closed 2024-03-06 by Bercon