SpectacularAI/sdk-examples #121: Error while starting Gsplat training via Nerfstudio

breezanil, 2024-01-13:
Hello,
I'm trying to get Gsplat file from a recording made with Spectacular Rec App on Iphone.
When I follow the tutorial, I get NERF properly from nerfstudio, but when I run the command ns-train gaussian-splatting --data \directory\ I encounter an error :

C:\Users\\anaconda3\envs\nerfstudio\lib\site-packages\torch\utils\cpp_extension.py:383: UserWarning: Error checking
compiler version for cl: [WinError 2] can't find the specified file

and then

File "C:\Users\\anaconda3\envs\nerfstudio\lib\site-packages\gsplat\cuda\_backend.py", line 56, in <module>
    from gsplat import csrc as _C
ImportError: cannot import name 'csrc' from 'gsplat' (C:\Users\breez\anaconda3\envs\nerfstudio\lib\site-packages\gsplat\__init__.py)

I updated Nerfstudio and reinstalled Pytorch but it still doens't work. Any idea ? 
Sorry I'm not really litterate in those Python/Cuda stuff...

Thanks, breezanil.

oseiskar, 2024-01-13:
Hi. Thanks for reporting. This is a bug in Nerfstudio/gsplat: https://github.com/nerfstudio-project/nerfstudio/issues/2685. Could you try the work-arounds suggested there, please, in particular, this one: https://github.com/nerfstudio-project/nerfstudio/issues/2685#issuecomment-1859660671

breezanil, 2024-01-13:
Thanks oseiskar, it worked.
I uninstalled gsplat, then reistalled gsplat 0.1.0 from source, as mentionned in the post.

closed 2024-03-15 by oseiskar