SpectacularAI/sdk-examples #128: SpectacularAI WARN: missing key frame in pose trail, skipping VIO constraint

breezanil, 2024-02-09:
Hello,

When I run the mapping tool, I sometimes get this message

warning: Lost tracking!
SpectacularAI WARN: missing key frame in pose trail, skipping VIO constraint

What does it mean exactly ? How could I avoid these ? It seems it disturb the mapping process.

Running on Windows 11
Dataset from IOS Spectacular Rec

oseiskar, 2024-02-09:
Our iOS parameters are currently a bit flaky and this will be improved in future SDK releases. The "lost tracking" message means that the system encountered and internal SLAM failure / reset and failed to produce a good reconstruction for this reason. You can ignore the other (internal) warning.

You may try again with different parameters (e.g., with a `--mono` command-line parameter for `sai-cli process`) or, if all else fails, try another recording. Avoiding fast movements, shooting objects at a very close distance and low lighting increase the risk of this happening.

breezanil, 2024-02-09:
Thanks a lot for these details.