SpectacularAI/sdk-examples #131: Could not open data.jsonl

khushalrs, 2024-02-28:
I just installed the spectacularAI using `pip install spectacularai[full]` 
When I run the `sai-cli  process` command it given me the following error

[image]

tjucaokun, 2024-03-04:
same problem. Have you solved it?

khushalrs, 2024-03-04:
Nope unable to resolve the error

kaatrasa, 2024-03-04:
Hey, sorry for the late reply, we have been quite busy.

I wasn't able to reproduce this issue with Windows 10 or Windows 11 using PowerShell.

1) Could you say which terminal you are using? (maybe miniconda's terminal?) and could you try with PowerShell and Git Bash instead.
2) Do the other [Python examples](https://github.com/SpectacularAI/sdk-examples/blob/main/python/oak/mapping_visu.py) work for you: For instance, `python mapping_visu.py --dataFolder /path/to/data`
3) Could be some issue with Python's `tempfile`, based on your error message. Could you check if the data.jsonl file mentioned in the ERROR message exists? We will try to get rid of it in next release.

kaatrasa, 2024-03-15:
This issue might have been fixed with release 1.31.0, just run `pip install spectacularai[full] --force`

closed 2024-06-18 by kaatrasa