SpectacularAI/sdk-examples #136: Couldn't find matching GLX visual

divisonofficer, 2024-06-20:
I'm using Ubuntu 22.04 on AMD64 platform with NVIDIA RTX 4060 (laptop). OpenGL version is 4.5 Messa 23.2.1-1ubuntu3.1~22.04.2

When i try to run ''mapping_vis.py'' via x11 connection,

```
Traceback (most recent call last):
  File "/home/cglab/project/spectacularAI-example/python/oak/mapping_visu.py", line 124, in <module>
    visualizer.run()
  File "/home/cglab/.local/lib/python3.10/site-packages/spectacularAI/cli/visualization/visualizer.py", line 569, in run
    self.__render(
  File "/home/cglab/.local/lib/python3.10/site-packages/spectacularAI/cli/visualization/visualizer.py", line 335, in __render
    self.__initDisplay()
  File "/home/cglab/.local/lib/python3.10/site-packages/spectacularAI/cli/visualization/visualizer.py", line 310, in __initDisplay
    else: pygame.display.set_mode((w, h), DOUBLEBUF | OPENGL)

pygame.error: Coudn't find matcing GLX visual
```

When i write my own code to open PyGame window in same condition, it works well.

kaatrasa, 2024-06-20:
Hey, thanks for letting us know. The source code for the Python visualizer is available [here](https://github.com/SpectacularAI/sdk/blob/main/python/cli/visualization/visualizer.py). Could you try if removing the multisampling fixes the issue for you?

oseiskar, 2024-06-24:
Also make sure your OpenGL drivers are correctly configured. "Mesa" drivers in use sounds like an issue in your setup. If you install the Nvidia driver and check that e.g., `glxgears` works (shows animated gears), then the SDK is likely to work too.