SpectacularAI/sdk-examples #145: How to use kinect sava real-time point cloud

GH-Fonic, 2024-11-21:
Thanks to the Spectacular AI team's work, I love ORBBEC's reconstruction of the point cloud in sai-record-orbbec, but I don't see how to save the data in ply or pcd, I don't see that in the documentation. I hope you can answer this question and thank the Spectacular AI team for their work :)

kaatrasa, 2024-11-22:
Hey, 

the `sai-record-orbbec` tool is mainly meant for recording with live preview and currently doesn't have an option to save the point cloud. Instead, I recommend you would use our `sai-cli process` commandline tool (installed with `pip install spectacularai[full]`).

To replay the dataset (with optimized parameters for offline use) and save the pointcloud, you can run:
`sai-cli process /path/to/recording /path/to/output/pointcloud.ply --preview3d`

You can see all the options with `--help`.

GH-Fonic, 2024-11-25:
thanks,I see. Thanks again for your work.
:)

closed 2024-11-25 by GH-Fonic